rsporny/homebridge-gpio-blinds #2: How to setup homebridge-gpio-blinds

migabc, 2017-08-05:
I would like to control my Velux (Shutters, Windows and blinds) using my RaspberryPI
I'm already using the homebridge with the homebridge-knx plugin
But I do not have an interface for my Velux.

Can you please describe how you did it?
Do you have a step-by-step guide?

Will I have to connect the GPIO pins from the RaspberryPi to the UP and DOWN contacts on the Velux remote control?
Is there a way to connect the GPIO pins to the Velux controller itself instead?

rsporny, 2017-08-09:
Hi, my blinds wires goes directly to the garage where I've RaspberryPi and the relay boards. Connection is pretty straightforward, just google for relay + raspberry.

In your case, you should send the signal to your Velux controller. Do you have any schematics or maybe pictures of it? How it communicates with the blinds?

migabc, 2017-08-09:
Hi,
Thanks for your reply.
I do not have schematics of the Velux controller. I don't think they publish this.
Some people have been able to connect the RPI pins directly to the Velux remote control:
https://github.com/ZoneMR/homebridge-gpio-blinds.
Not sure which pins to connect.

migabc, 2017-08-16:
I believe this is exactly what I need (as it connects to a remote control terminals that is exactly like mine)
https://github.com/ZoneMR/homebridge-gpio-blinds
How do I install this plugin?
There's no installation instructions in the above link and it's not a npm
Any help is welcome.

rsporny, 2017-08-18:
You need to download the code directly to your homebridge plugin dir. Just like you would start developing a new plugin. Then you need to run homebridge again and it should work. For more info go to https://github.com/nfarina/homebridge#plugin-development

closed 2017-11-20 by rsporny