rsporny/homebridge-gpio-blinds #3: Relay management

kiondi, 2017-08-10:
Hello
The plugin works fine with homebridge.
How can I do to reverse the GPIO exit when it work?
The plugin set the port **HIGH**as default, when the blind stop, and **LOW** when the blind work. 
Thanks

kiondi, 2017-08-11:
I need trigger the relay by the HIGH level

rsporny, 2017-08-18:
Hi, sure I will make the active gpio configurable in a couple of days.

hyperbart, 2017-11-20:
Might be related to my issue to. Afaik we need a HIGH to start the blinds.

hyperbart, 2017-11-20:
I saw you pushed, will look through code later on, super work, thanks!

Just out of curiosity: do you mind sharing which motor you have for your blinds and how you have wired them to your relay board (NO/NC?). Because both @martijndierckx and I were pretty surprised to see that you need an inverted open to control your motor?

rsporny, 2017-11-20:
@hyperbart I've got relay module as in README [here](https://botland.com.pl/przekazniki/6941-szesnastokanalowy-modul-przekaznikow-rm13-12v-z-izolacja-optoelektroniczna-10a125vac.html?search_query=MOD-06941&results=1). All the wires from the roller blinds goes directly to garage where this whole setup is connected with raspberry. I don't have any switches near windows, all of this is controlled via iOS Home app only.

[image]

rsporny, 2017-11-20:
@kiondi sorry for taking it so long. Fixed in [v1.0.1](https://github.com/rsporny/homebridge-gpio-blinds/tree/v1.0.1).

closed 2017-11-20 by rsporny

hyperbart, 2017-11-20:
Works perfect!

You're gonna laugh a little bit but I found a new "enhancement" 😄 .

A dual-controlled blind (a blind with a fixed switch and the homebridge-gpio-blinds) does not report back a state, so you could get a mismatched object in HomeKit where it thinks it is closed or open while it's the opposite. The guys at homebridge-rasppi-gpio-garagedoor have solved this by including a reed-switch/magnet-switch which signals is the garagedoor is open or closed.