rsporny/homebridge-gpio-blinds #4: Explanation pins HIGH at boot

hyperbart, 2017-11-20:
Hi

A friend of mine and I have been looking at this plugin but we do have a question about why you are booting/starting the plugin with both the up and down pin as HIGH/1.

If I would hook this up to my blind motor I think I would break something since I'm sending 230V in both directions at the same time.

Why is that?

Regards

Bart

rsporny, 2017-11-20:
Hi @hyperbart 
I'm glad you ask, this is in my mind for a couple of months now (as you've seen in issue #3). My relay is enabled (turned on) by LOW state, but this should be configurable. I'll fix this next week.

hyperbart, 2017-11-20:
@rsporny . Aha explains a little bit.

Would be nice yeah, because the house where I want to implement your plugin has blinds/shutters which are controlled by a "turn" knob, something like this:

[image]

And all that knob does is "patching through" the hot wire/phase wire to the motor.

Looking forward to it mate, cheers! Looks like a nice house btw! ;)

rsporny, 2017-11-20:
Thanks :) I'm fixing it right now so stay tuned.

hyperbart, 2017-11-20:
Be advised, a friend of mine issued a pull request, might take a look at it since it accomplishes what you want to achieve?

rsporny, 2017-11-20:
Fixed in [v1.0.1](https://github.com/rsporny/homebridge-gpio-blinds/tree/v1.0.1). If you have further questions feel free to reopen.

closed 2017-11-20 by rsporny