rsporny/homebridge-gpio-blinds #6: Report state of blinds (closed) via button/reedswitch

hyperbart, 2017-11-20:
If a blind is controlled by the plugin AND an external switch (like a wall switch) the state of the object in HomeKit might get outdated or out of sync.

Suggestion: implement fix with one or two (closed is the most important one I think) reedswitches like the guys at homebridge-rasppi-gpio-garagedoor

rsporny, 2017-12-06:
Yes, that's on my roadmap. I'll be doing similar plugin to `homebridge-rasppi-gpio-garagedoor` but based on rpio lib (no root needed). When it's done, I'll use the reed switch functionality in blinds as well.

morphix9, 2018-01-28:
@rsporny I just tried your plugin. It works like a charm, thanks for sharing it!
How is your progress on the state of each blind when an external button is wired to the relay?

rsporny, 2018-02-03:
I'm working on it on a second branch. New version should be released soon. It will have two additional pins to configure with reed switches, one for closed and the other for open state, and will synchronize `currentPosition` and `targetPosition` accordingly. If none of the reed switches meets its position I'll probably set the position to 50% - this may be a little out of sync but I'm afraid there is no better solution with such setup. Just be aware that sending signal to your blinds for too long may burn them if they do not have limit/end switch.

rsporny, 2018-05-17:
@morphix9 @hyperbart I've just published v1.1.0 which resolves the issue. See readme for config details. Please test it and reopen if you spot some issue 🐛

closed 2018-05-17 by rsporny