rsporny/homebridge-gpio-blinds #7: Plugin takes the target position wrong

radebebek, 2018-01-11:
Plugin takes target position wrong when I slow move slider.
I moved slider from 0% to 50% slow, but plugin take some position in middle (for example 20%)
But when I exit application HOME and after that turn on application. Show Position in correct 20%.
After a few such steps, blinds never come to 100% because duration is wrong, blinds near open maybe 95% physical but in application 100%.
Can you fix it. Maybe that plugin wait some time (3 s) and after that take Target Position.

Rade

rsporny, 2018-01-11:
Yes, this is a known issue. To be honest, I get used to it and at first started to move the slider quickly, now I've got some automations and I don't open/close blinds manually at all. But as somebody else is using it, I'm definitely take a closer look to it.

radebebek, 2018-01-12:
That would be great, I can help you if I succes change code in my environment.
I'm currently working on my own plugin (https://github.com/radebebek/homebridge-sonoff-blinds) , which uses http requests. It's based on your code. But I'm working with the sonoff device which  working on wirelles.

radebebek, 2018-02-02:
Some progress?

rsporny, 2018-05-17:
Hey, I've just published new version, looked at the code and got some ideas to push this ticket forward. Stay tuned ⏳

erokoder, 2018-05-17:
@rsporny  I will try new code.

rsporny, 2018-05-18:
v1.1.1 resolves the bug. Please verify if you can.

closed 2018-05-18 by rsporny