rsporny/homebridge-gpio-blinds #8: Move blinds with external switches

morphix9, 2018-03-26:
Hey,
your plugin is great, I tried it out and it works great. 

But I also have external (physical) switches to move each blind up/down. This is causing an off-sync with the state remembered by the homebridge. How hard do you think would it be to assign each blind also two input pins that would move them up/down via homebridge. Then the state could be in sync without any position sensors. Would that be possible?

rsporny, 2018-05-17:
Hi,
it's possible, however it would require from me more research about it. Given my JS skills and lack of free time I cannot say when I'll do it. Let's leave this ticket open and rename title. Maybe someone else would like to help with this.

morphix9, 2018-05-17:
Cool, no worries. Do you think it's even possible to do this with the 'rpio' library? 'onoff' library seams better, as it has interrupt detection and rpio not, or am I wrong?

rsporny, 2018-05-17:
thanks, I'll take a closer look into it.

morphix9, 2018-09-03:
@rsporny  any change to test rpio.poll for input pins (as triggers for up/down)?

EnriqueSubias, 2019-05-22:
Why don’t you install switchs that control the blind via HomeKit, like the Philips ones. Or maybe even via ssh. (Although it won’t update in HomeKit)?

morphix9, 2019-05-23:
> Why don’t you install switchs that control the blind via HomeKit, like the Philips ones. Or maybe even via ssh. (Although it won’t update in HomeKit)?

Yes, that's the point to have it synced with HomeKit.

chrisremie, 2020-12-07:
I would also like this feature.

And thanks for making this plugin!