rsporny/homebridge-gpio-blinds #14: Error on installation

lordchima, 2018-10-06:
Hey,

Youre plugin seems really cool. Its exactly what i needed.
Sadly i get an error.

[2018-10-6 13:35:21] ====================
[2018-10-6 13:35:21] ERROR LOADING PLUGIN homebridge-gpio-blinds:
[2018-10-6 13:35:21] Error: Could not locate the bindings file. Tried:
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/build/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/build/Debug/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/build/Release/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/out/Debug/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/Debug/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/out/Release/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/Release/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/build/default/rpio.node
 → /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/compiled/8.9.1/linux/arm/rpio.node
    at bindings (/usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/bindings/bindings.js:93:9)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio/lib/rpio.js:17:34)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
    at Module.require (module.js:579:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-gpio-blinds/index.js:2:12)
[2018-10-6 13:35:21] ====================

Maybe you could help me.

lordchima, 2018-10-11:
No one?

rsporny, 2018-10-11:
Hey, what npm and node version do you have? What raspberry model are you using?

Torac2, 2018-10-14:
Hey sorry for the long wait my old account i forgot the password haha.
I got an raspberry pi 3B npm and node version how do i see those?

Torac2, 2018-10-16:
> Hey, what npm and node version do you have? What raspberry model are you using?

npm version 6.4.1
node version 9.9.1

Torac2, 2018-10-21:
Yo @rsporny ?

rsporny, 2018-10-22:
Please try downgrading npm to 3.10.3 and node to 6.6.0.

rsporny, 2018-10-22:
I'll try the newest versions soon, but if the old doesn't work for you, please remove node-modules dir and install plugin again.

Torac2, 2018-10-26:
Thanks for the help the error is gone! :)
Sadly im now getting this error.

/usr/lib/node_modules/homebridge/lib/api.js:64
      throw new Error("The requested accessory '" + name + "' was not registered by any plugin.");
      ^

Error: The requested accessory 'Blinds' was not registered by any plugin.
    at API.accessory (/usr/lib/node_modules/homebridge/lib/api.js:64:13)
    at Server._loadAccessories (/usr/lib/node_modules/homebridge/lib/server.js:264:42)
    at Server.run (/usr/lib/node_modules/homebridge/lib/server.js:86:38)
    at module.exports (/usr/lib/node_modules/homebridge/lib/cli.js:40:10)
    at Object.<anonymous> (/usr/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Function.Module.runMain (module.js:605:10)
    at startup (bootstrap_node.js:158:16)
    at bootstrap_node.js:575:3

Config:

{
    "accessory": "Blinds",
    "name": "Sleeproom",
    "pinUp": 5,
    "pinDown": 11,
    "durationUp": 13000,
    "durationDown": 13000,
    "durationOffset": 1000,
    "pinClosed": 17,
    "pinOpen": 18,
    "activeLow": false,
    "reedSwitchActiveLow": false
  },

rsporny, 2018-11-14:
Sorry, did you manage to fix it? Probably you still don't have plugin installed. What you get when you run this `npm install -g homebridge-gpio-blinds`?

closed 2019-04-17 by rsporny