rsporny/homebridge-gpio-blinds #15: Servo Motor

thgcst, 2018-11-07:
Is it possible to use a Servo motor such as MG995 modified to continuous rotate? How should be the connection for Up and Down pins?

hennessyevan, 2018-11-07:
I actually initially was going to use a servo, the one I had wasn't strong enough for my blinds but it was easy enough to modify this project for that usage. You will need to change some of the code to work properly with PWM (Pulse Width Modulation) which is what servos run on.

I began to modify this module for use with my servo but then I abandoned it and I can't find that version anymore 😞  but you can take a look at [this](https://github.com/hennessyevan/homebridge-servo-blinds/blob/master/index.js) which may shed some light on PWM for servos in general. You just need to change the `rpio.write()` functions to make use of PWM.

If you get stuck I may be able to help

thgcst, 2018-11-09:
Thanks a lot for your help.
I don't know exactly how to make it work, but i was expecting something like this code here. Set a time for open and close the blinds and the code does the rest such as opening percentage. The thing is, I got the MG995 Servo based on [this](https://www.youtube.com/watch?v=8bcYB-0bctE&t=675s), but expecting to use this GPIO-BLINDS code with the MG995 Servo.

Another problem I got is: this Servo is 180º limitaded, but with [this video](https://www.youtube.com/watch?v=8bcYB-0bctE&t=675s) I realized that there's a way to make it continuous rotate. But actually I am very amateur on programming and can't make everything work.

rsporny, 2018-11-14:
Since rpio supports PWM it seems easy enough. If you manage to do some sort of proof of concept, please share the code and I'll think how to support it in this plugin.