rsporny/homebridge-gpio-blinds #21: Installation problem: Failed at the rpio@0.9.23 install script.

BabbaBernd, 2020-01-05:
I have problems with the installation of rpio ... at the end of the long long log file i got the following:
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.19.75-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/homebridge-gpio-blinds/node_modules/rpio
gyp ERR! node -v v12.14.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rpio@0.9.23 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rpio@0.9.23 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

franzvb, 2020-03-18:
have the exact same problem. My research suggests the plugin needs to be updated... is that correct? anybody knows how to do that?

rsporny, 2020-09-27:
thanks for raising the issue.
fixed in v1.1.3 and released to npmjs

closed 2020-09-27 by rsporny