rsporny/homebridge-gpio-blinds #24: durationOffset doesn't seem to do anything

chrisremie, 2021-11-17:
I have set durationOffset, but in the log file the duration is just the value from durationUp/Down.

rsporny, 2021-11-17:
hi @chrisremie,
the code is:
`if (this.durationOffset && (this.targetPosition == 0 || this.targetPosition == 100)) this.duration += this.durationOffset;`
so durationOffset only extend duration when you want to fully close/open blinds. I'll modify description so it's more obvious. Does this solve your problem?

chrisremie, 2021-11-17:
Hi,

Thanks for your fast reply.

This is my config:
  {
            "accessory": "Blinds",
            "name": "Salone",
            "pinUp": 4,
            "pinDown": 5,
            "durationUp": 20000,
            "durationDown": 20000,
            "durationOffset": 4000,
            "activeLow": true
        },

And this is from the log:
[17/11/2021, 10:41:29] [Salone] Setting target position to 100
[17/11/2021, 10:41:29] [Salone] Moving up. Duration: 20000 ms.
[17/11/2021, 10:41:49] [Salone] Successfully moved to target position: 100

[17/11/2021, 10:42:34] [Salone] Setting target position to 0
[17/11/2021, 10:42:34] [Salone] Moving down. Duration: 20000 ms.
[17/11/2021, 10:42:54] [Salone] Successfully moved to target position: 0

So somehow it does not add the Offset

Chris.

rsporny, 2021-11-17:
I see. The logs are not taking into account durationOffset, I'll add new log entry for that. However, the code should work as expected, could you confirm by checking for how long relay switch is on?

chrisremie, 2021-11-17:
It is not adding the offset.
I added a really big offset, and it is still not closing completely

rsporny, 2021-11-17:
Thanks. Could you please confirm the version you are using?

chrisremie, 2021-11-17:
V1.1.3

rsporny, 2021-11-17:
Thanks, I'll investigate.

chrisremie, 2021-11-22:
index.js
line 162
 if (this.durationOffset && (this.targetPosition == 0 || this.targetPosition == 100)) this.duration += this.durationOffset;

this.duration should be duration.